ZJU-FAST-Lab/ego-planner-swarm #3: Manually trigger replanning

Kavin-Kailash, 2021-03-14:
Hello. Thanks for the great work.

I am interested in using this package in conjunction with a custom exploration planner, which provides path segments at some fixed interval of time.

I can see that the ego_replan_fsm file can be modified to accept waypoints from a external topic, however I am unclear as to how to make the planner track a new set of waypoints, while it's already executing the previous one.

Thanks in advance!!

bigsuperZZZX, 2021-03-27:
Sorry for replying so late since I'm super busy recently!
It's easy to achieve that. All you need is to 
First: change the "flight_type" to "1" in [run_in_sim.launch, line 60](https://github.com/ZJU-FAST-Lab/ego-planner-swarm/blob/5ddc525bfb83f6e015c630e44223c6296c755376/src/planner/plan_manage/launch/run_in_sim.launch#L60) to enable waypoint input when flying.
Second: noticeing the callback function in [ego_replan_fsm.cpp, line 64](https://github.com/ZJU-FAST-Lab/ego-planner-swarm/blob/5ddc525bfb83f6e015c630e44223c6296c755376/src/planner/plan_manage/src/ego_replan_fsm.cpp#L64), you need to publish that topic which is a waypoint to make the drone to follow.

bigsuperZZZX, 2021-04-04:
Hello, if you have no other questions, please close this issue.

closed 2021-04-04 by Kavin-Kailash