ZJU-FAST-Lab/ego-planner-swarm #5: Terminal point of the current trajectory is in obstacle, skip this planning

robin-shaun, 2021-04-01:
Hello, I sometimes meet continuous warning `terminal point of the current trajectory is in obstacle, skip this planning`  in my PX4 SITL Simulation and the UAV will stop. I think this will happen when there's no solution to reach the goal, but indeed when this happens no obstacle is in front of the UAV.

In most cases, ego_planner works quite well. Thanks for the wonderful work.

[image]

[image]

bigsuperZZZX, 2021-04-04:
If no obstacle is in front of the UAV while this error keeps reporting, then maybe the given goal is outside the global map, or the goal is insider the inflated obstacles. Please check that.

robin-shaun, 2021-04-04:
Is the global map size given by `  <arg name="map_size_x"/>
  <arg name="map_size_y"/>
  <arg name="map_size_z"/>` in  [here](https://github.com/ZJU-FAST-Lab/ego-planner-swarm/blob/master/src/planner/plan_manage/launch/run_in_sim.launch)?

bigsuperZZZX, 2021-04-04:
Yes. The map region is [-map_size_x/2, map_size_x/2], [-map_size_y/2, map_size_y/2], [[ground_height](https://github.com/ZJU-FAST-Lab/ego-planner-swarm/blob/5ddc525bfb83f6e015c630e44223c6296c755376/src/planner/plan_manage/launch/advanced_param.xml#L97), map_size_z-ground_height].

robin-shaun, 2021-04-04:
OK, I'll try it.

bigsuperZZZX, 2021-04-06:
Hello, if you have no other questions, please close this issue.

closed 2021-04-06 by robin-shaun