ZJU-FAST-Lab/ego-planner-swarm #6: Yaw speed and height constraints

robin-shaun, 2021-04-01:
Hello, I see that there are constraints for velocity `max_vel`and acceleration `max_acc` in the launch file. Is there any way to constraint the yaw speed because the large yaw speed may cause VIO failure.

Moreover, I see `grid_map/virtual_ceil_height`,  `grid_map/visualization_truncate_height` and `map_size_z` in the launch files. But it seems that they cannot limit the z position setpoint.

bigsuperZZZX, 2021-04-04:
First question: the way to constrain yaw speed is trivial, which can be changed [here](https://github.com/ZJU-FAST-Lab/ego-planner-swarm/blob/5ddc525bfb83f6e015c630e44223c6296c755376/src/planner/plan_manage/src/traj_server.cpp#L74).

Second question: is the "setpoint" refer to the goal you give? There is no way to limit it, you have to take of it yourself.

robin-shaun, 2021-04-04:
Thanks a lot. The setpoint is the trajectory given by ego_planner. For example, generally in the indoor simulation, there is no ceiling. If there is no height limit, ego_planner may give a trajectory over the wall, which is not the same as the real world.

bigsuperZZZX, 2021-04-04:
Do you mean that the planner outputs some setpoints above the "grid_map/virtual_ceil_height"? It should not happen. 
First, make sure the "grid_map/virtual_ceil_height" is lower than the "map_size_z".
Then, maybe you can set "grid_map/visualization_truncate_height" higher than the "grid_map/virtual_ceil_height" to see how the virtual ceil looks like by visualizing the inflated grid map and check if the planned trajectory can break the ceiling limit.

robin-shaun, 2021-04-04:
Ok, I 'll try it~

bigsuperZZZX, 2021-04-06:
Hello! If you have no other questions, please close this issue.

closed 2021-04-06 by robin-shaun