ZJU-FAST-Lab/ego-planner-swarm #9: Fix for missing MultiOccupancyGrid.h

madasebrof, 2021-05-18:
If you try to build on a clean machine (tested on 18.04 & 20.04), you'll get an error due to a missing header file `multi_map_server/MultiOccupancyGrid.h`.

To fix this, edit the file:

```
src/uav_simulator/Utils/multi_map_server/CMakeLists.txt
```

Comment out the executable generation code per below:

```
## Declare a cpp executable
# COMMENT OUT FOR FIRST COMPILE PASS
# add_executable(multi_map_visualization src/multi_map_visualization.cc)

## Add cmake target dependencies of the executable/library
## as an example, message headers may need to be generated before nodes
# COMMENT OUT FOR FIRST COMPILE PASS
# add_dependencies(multi_map_visualization multi_map_server_messages_cpp)

## Specify libraries to link a library or executable target against
# COMMENT OUT FOR FIRST COMPILE PASS
# target_link_libraries(multi_map_visualization
#    ${catkin_LIBRARIES}
#    ${ARMADILLO_LIBRARIES}
#   pose_utils
# )
```

Then compile first pass, e.g. 

```
catkin_make -j1
```

Now, you'll have `devel/include/multi_map_server/MultiOccupancyGrid.h` available.

Then, uncomment the file:

```
## Declare a cpp executable
# COMMENT OUT FOR FIRST COMPILE PASS
add_executable(multi_map_visualization src/multi_map_visualization.cc)

## Add cmake target dependencies of the executable/library
## as an example, message headers may need to be generated before nodes
# COMMENT OUT FOR FIRST COMPILE PASS
add_dependencies(multi_map_visualization multi_map_server_messages_cpp)

## Specify libraries to link a library or executable target against
# COMMENT OUT FOR FIRST COMPILE PASS
target_link_libraries(multi_map_visualization
   ${catkin_LIBRARIES}
   ${ARMADILLO_LIBRARIES}
   pose_utils
)
```

and recompile:

```
catkin_make -DCMAKE_BUILD_TYPE=Release -j1
```

Should work now!

bigsuperZZZX, 2021-05-18:
Thank you for this instruction!
Hoping it will help someone in need.

closed 2021-05-18 by bigsuperZZZX

Zoe20224, 2022-05-01:
really useful! Your suggestion helped me a lot! Thx!