ZJU-FAST-Lab/ego-planner-swarm #10: “catkin_make -j1”

ZK-SJTU, 2021-05-31:
function catkin_make -j1，an  error occurred.

/usr/local/include/eigen3/Eigen/src/Core/Assign.h:55:30: note:   no known conversion for argument 1 from ‘double’ to ‘const Eigen::MatrixBase<Eigen::IndexedView<Eigen::Matrix<double, -1, -1>, double, double> >&’
planner/traj_utils/CMakeFiles/traj_utils.dir/build.make:86: recipe for target 'planner/traj_utils/CMakeFiles/traj_utils.dir/src/polynomial_traj.cpp.o' failed
make[2]: *** [planner/traj_utils/CMakeFiles/traj_utils.dir/src/polynomial_traj.cpp.o] Error 1
CMakeFiles/Makefile2:5925: recipe for target 'planner/traj_utils/CMakeFiles/traj_utils.dir/all' failed
make[1]: *** [planner/traj_utils/CMakeFiles/traj_utils.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j1" failed

bigsuperZZZX, 2021-06-01:
I have never encountered such problems. Seems that there is something wrong with the compiling environment.

USTfgaoaa, 2021-06-15:
@ZK-SJTU Try to delete all build and devel files and re-compile the whole workspace with -j1.

closed 2021-06-15 by USTfgaoaa