ZJU-FAST-Lab/ego-planner-swarm #12: <multi_map_server/MultiOccupancyGrid.h> not found

aswinnahrendra, 2021-06-07:
The `<multi_map_server/MultiOccupancyGrid.h>` file is not included in the repo, leading to failed build. Thanks~

bigsuperZZZX, 2021-06-09:
This .h file is generated from a .msg file "MultiOccupancyGrid.msg", therefore, you should use only one thread by "catkin_make -j1" when compiling to let the ROS build tools compile the code following the right order.

chenshun1997, 2021-06-15:
fatal error: multi_map_server/MultiOccupancyGrid.h: 没有那个文件或目录
    5 | #include <multi_map_server/MultiOccupancyGrid.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

closed 2021-06-15 by USTfgaoaa

aphasiayc, 2022-06-02:
Had the same problem. Fixed it by changing [this line](https://github.com/ZJU-FAST-Lab/ego-planner-swarm/blob/8da8a01e7d01a9977306aaf2922cf85a1a3de71e/src/uav_simulator/Utils/multi_map_server/CMakeLists.txt#L167) to
```cmake
add_dependencies(multi_map_visualization multi_map_server_generate_messages_cpp)
```

bigsuperZZZX, 2023-03-16:
Use -j1 only if you encounter that error. 
catkin_make works for most of the users.