ZJU-FAST-Lab/ego-planner-swarm #13: double free or corruption (out)

namanyyy, 2021-07-14:
double free or corruption (out)
[drone_2_ego_planner_node-13] process has died [pid 16724, exit code -6, cmd /home/naman/Program/ego-planner-swarm/devel/lib/ego_planner/ego_planner_node ~odom_world:=/drone_2_visual_slam/odom ~planning/bspline:=/drone_2_planning/bspline ~planning/data_display:=/drone_2_planning/data_display ~planning/broadcast_bspline_from_planner:=/broadcast_bspline ~planning/broadcast_bspline_to_planner:=/broadcast_bspline ~grid_map/odom:=/drone_2_visual_slam/odom ~grid_map/cloud:=/drone_2_pcl_render_node/cloud ~grid_map/pose:=/drone_2_pcl_render_node/camera_pose ~grid_map/depth:=/drone_2_pcl_render_node/depth __name:=drone_2_ego_planner_node __log:=/home/naman/.ros/log/2fe9fc80-e44c-11eb-8017-ccf9e4e8181d/drone_2_ego_planner_node-13.log].
log file: /home/naman/.ros/log/2fe9fc80-e44c-11eb-8017-ccf9e4e8181d/drone_2_ego_planner_node-13*.log

请问这是什么情况呢

bigsuperZZZX, 2021-08-15:
这是内存读写错误，但是这段长长的报错只是该节点的启动指令而已，啥有用的信息都没有，这种内存错误通常只能用户自己取查了

closed 2021-08-16 by ZhepeiWang

ZJU-kingdong, 2021-08-16:
是不是地图尺寸给太大了

bigsuperZZZX, 2021-08-16:
是有可能

DavidZhaoP, 2022-05-08:
> 是有可能
你好，我也遇到这个问题，并且问题
定位到ego_planner中的PolynomialTrajPolynomialTraj::minSnapTraj(constEigen::MatrixXd&Pos, constEigen::Vector3d&start_vel, constEigen::Vector3d&end_vel, constEigen::Vector3d&start_acc, constEigen::Vector3d&end_acc, constEigen::VectorXd&Time) 函数中Eigen::MatrixXd R = C * A.transpose().inverse() * Q * A.inverse() * Ct;中出了问题，并且可以在这行代码前面自己开辟相同的矩阵Eigen::MatrixXd R2= C * A.transpose().inverse() * Q * A.inverse() 可以通过，意味着只要不乘最后的Ct矩阵就可以运行.......

shenyuan411, 2023-08-20:
> 

请问这个问题现在解决了吗

Abonaventure, 2024-02-28:
It's normal when I compiled and run ego-planner, but I come across the same problem when I  compiled and run ego-swarm.(catkin_make is OK，catkin build failed) .To solve ego-swarm problem, I tried downscale map size, check the vecfree() function and change "Eigen::MatrixXd R = C * A.transpose().inverse() * Q * A.inverse() * Ct", but the problem is still the same. Do anyone know how to solve this problem?

AIRcrafts, 2025-04-25:
You can try adding a `set(CMAKE_CXX_FLAGS "-std=c++11 -O2 ${SSE_FLAGS} -g -march=native") ` in the src/planner/traj_utils/CMakeLists.txt.