ZJU-FAST-Lab/ego-planner-swarm #20: no consideration of drone odometry while replanning

shubham-shahh, 2021-11-10:
Hi,
Why the planner does not consider the current odometry of the UAV in the replanning step
if the UAV is stuck at a point the planner traj keeps moving forward even if UAV if the UAV is stuck.

thanks

bigsuperZZZX, 2021-11-10:
Planning starts from a point on the current trajectory is to guarantee command continuity, therefore improving smoothness.

The UAV is guaranteed to not get away from the trajectory in our trajectory-planning-following framework. Therefore some external, forced stop is not considered in the current version. However, the [FSM(finite state machine)](https://github.com/ZJU-FAST-Lab/ego-planner-swarm/blob/master/src/planner/plan_manage/src/ego_replan_fsm.cpp) has the potential to tackle such situations. You can just change the planner state to [GEN_NEW_TRAJ](https://github.com/ZJU-FAST-Lab/ego-planner-swarm/blob/2b7a7fa2300d37048d8b8e6d67bb243c7883f8f8/src/planner/plan_manage/src/ego_replan_fsm.cpp#L508) when significant trajectory tracking deviation is detected. In this state, the planner plans a trajectory from the current odometry readings.

shubham-shahh, 2021-11-10:
Hi, thanks for the response, but detecting a trajectory tracking deviation is difficult, that's why I want to try to combine the odometry in the planner. for a single UAV, I found following places where modification could be made

[1](https://github.com/ZJU-FAST-Lab/ego-planner-swarm/blob/2b7a7fa2300d37048d8b8e6d67bb243c7883f8f8/src/planner/plan_manage/src/ego_replan_fsm.cpp#L553), [2](https://github.com/ZJU-FAST-Lab/ego-planner-swarm/blob/2b7a7fa2300d37048d8b8e6d67bb243c7883f8f8/src/planner/plan_manage/src/ego_replan_fsm.cpp#L649) along with start_vel_ and start_accel_

please current me if there are additional places or If I pointed incorrect places.

also In your opinion, what are your thoughts what could be the possible downsides of adding odometry?

thanks

bigsuperZZZX, 2021-11-10:
Modifying [this](https://github.com/ZJU-FAST-Lab/ego-planner-swarm/blob/2b7a7fa2300d37048d8b8e6d67bb243c7883f8f8/src/planner/plan_manage/src/ego_replan_fsm.cpp#L649) is strongly not recommended since harming the continuity will lead to severe results.

In fact, detecting a tracking error is easy, you only need to compare the position in odometry topic with the [command topic](https://github.com/ZJU-FAST-Lab/ego-planner-swarm/blob/2b7a7fa2300d37048d8b8e6d67bb243c7883f8f8/src/planner/plan_manage/src/traj_server.cpp#L240) and set a difference threshold.

shubham-shahh, 2021-11-10:
> Modifying [this](https://github.com/ZJU-FAST-Lab/ego-planner-swarm/blob/2b7a7fa2300d37048d8b8e6d67bb243c7883f8f8/src/planner/plan_manage/src/ego_replan_fsm.cpp#L649) is strongly not recommended [...]

Okay, I'll try that. thanks for all the help

closed 2021-12-09 by bigsuperZZZX