ZJU-FAST-Lab/ego-planner-swarm #23: Ego planner map size constraints

shubham-shahh, 2021-12-08:
Hi, @bigsuperZZZX 
The planner is constrained by the map size that is initially given. [here](https://github.com/ZJU-FAST-Lab/ego-planner-swarm/blob/8da8a01e7d01a9977306aaf2922cf85a1a3de71e/src/planner/plan_env/src/grid_map.cpp#L88) the size of the buffer becomes large which limits the maximum size.

are there any approaches to make the planner free from the map size constraints?

thanks

bigsuperZZZX, 2021-12-09:
Using a [3D-circular buffer](https://arxiv.org/abs/1703.01416) can solve this. You can code by yourselves or wait for our brand new planner along with a ring map which will be released in about 2 months.

shubham-shahh, 2021-12-09:
> Using a [3D-circular buffer](https://arxiv.org/abs/1703.01416) can solve this. You can code by yourselves or wait for our brand new planner along with a ring map which will be released in about 2 months.

okay, thanks for the update

shubham-shahh, 2021-12-09:
Hi @bigsuperZZZX,
is it necessary for[ this](https://github.com/ZJU-FAST-Lab/ego-planner-swarm/blob/8da8a01e7d01a9977306aaf2922cf85a1a3de71e/src/planner/plan_env/src/grid_map.cpp#L88) vector to store double type data? is this much amount of precision required?

I found the range of values stored in the ` md_.occupancy_buffer_` variable ranges from -2.00243 to 2.197, can you please explain me what does this values translate to? I thought it must be storing probability ranging from 0-1

thanks

bigsuperZZZX, 2021-12-10:
It is [log-odds](https://en.wikipedia.org/wiki/Logit) representation not the original probability value.

And using double is not necessary indeed.

shubham-shahh, 2021-12-10:
> 

thanks for the information.

closed 2021-12-10 by shubham-shahh

shubham-shahh, 2022-05-10:
> Using a [3D-circular buffer](https://arxiv.org/abs/1703.01416) can solve this. You can code by yourselves or wait for our brand new planner along with a ring map which will be released in about 2 months.

Hi @bigsuperZZZX, I am curious if there are any updates to the latest paper

Thanks

bigsuperZZZX, 2022-05-10:
You would like to refer to this [repo](https://github.com/ZJU-FAST-Lab/EGO-Planner-v2)