ZJU-FAST-Lab/ego-planner-swarm #27: issue with path planning

shubham-shahh, 2022-03-04:
Hi, @bigsuperZZZX 
I am having an issue with tall and narrow obstacles, The planner tries to search path from top and bottom rather than finding pats from sides (favourable for this case). How can solve this

all your help is appreciated

thanks

bigsuperZZZX, 2022-03-05:
Could you paste a picture here about what your obstacles look like?

shubham-shahh, 2022-03-05:
Hi @bigsuperZZZX , here's a picture of the obstacle and the occupancy grid[](url
[image]
)

bigsuperZZZX, 2022-03-05:
That's because the obstacle sensor only senses a limited region.
From the mapped obstacle in the right figure, the produced trajectory is reasonable as it chooses the shortest path to fly through because the drone takes the unknown part as free.

shubham-shahh, 2022-03-06:
> That's because the obstacle sensor only senses a limited region. From the mapped obstacle in the right figure, the produced trajectory is reasonable as it chooses the shortest path to fly through because the drone takes the unknown part as free.

@bigsuperZZZX , That's correct, but, it will keep moving upwards as it maps the obstacle, but it is not good for such kind of obstacles, what are the possible solutions, to tackle such obstacles.
I have mounted a lidar with 90 degree vertical FOV for obstacle detection

thanks

bigsuperZZZX, 2022-03-06:
It's a complicated problem to tackle obstacles outside the sensing region. 
There are different candidate strategies, such as predicting obstacle shape using neural networks [(paper)](https://ieeexplore.ieee.org/abstract/document/9636333/), adjusting trajectory to see more known places [(paper)](https://ieeexplore.ieee.org/document/9720920), or just using wide range sensors like fisheye cameras [(paper)](https://onlinelibrary.wiley.com/doi/abs/10.1002/rob.21946), etc.

shubham-shahh, 2022-03-07:
> It's a complicated problem to tackle obstacles outside the sensing region. There are different candidate strategies, such as predicting obstacle shape using neural networks [...]

Hi, Thanks for the response. I'll look into them

closed 2022-03-07 by shubham-shahh