ZJU-FAST-Lab/ego-planner-swarm #30: when you avoid obstacles, how do you ensure that the plane will not collide

001zwzwzw, 2022-04-06:
hello!I want to know the method that when you avoid obstacles, how do you ensure that the plane will not collide

001zwzwzw, 2022-04-08:
@ZhepeiWang

001zwzwzw, 2022-04-08:
@ZhepeiWang @USTfgaoaa i need your help!Thanks !

ZhepeiWang, 2022-04-08:
@bigsuperZZZX

bigsuperZZZX, 2022-04-08:
The plane itself does not need to be strictly collision-free. It just provides the information to push the only control point that generates this plane away from obstacles. Whether the final trajectory is collision-free is guaranteed by a post-check procedure.

001zwzwzw, 2022-04-09:
> The plane itself does not need to be strictly collision-free. It just provides the information to push the only control point that generates this plane away from obstacles. Whether the final trajectory is collision-free is guaranteed by a post-check procedure.

 @bigsuperZZZX  Thanks very much ! than,i want to ask the problem that every trajectories of drone 0~9 are known other drones?For example the trajectory of drone 0 is shared with other drones like drone 1~9?how do you achieve them in procedure

001zwzwzw, 2022-04-09:
@bigsuperZZZX Thanks very much ! then,i want to ask the problem that every trajectories of drone 0-9 are known other drones?For example the trajectory of drone 0 is known by other drones like drone 1-9? how do you achieve them in procedure

bigsuperZZZX, 2022-04-09:
Everyone knows others' trajectories via a wireless broadcasting network.

001zwzwzw, 2022-04-09:
@bigsuperZZZX  Wether this method will bring a communication pressure in real test of UAV-swarm ?thanks!

bigsuperZZZX, 2022-04-09:
It's hard to tell since wireless communication quality is affected by many factors. 
According to our tests, a swarm of ten drones enjoys the WiFi connection.
We haven't tested more agents as we have 10 drones in total.

001zwzwzw, 2022-04-09:
@bigsuperZZZX  ok I learn more knowledge from you! Thank you, senior brother!
about hardware equipment,do you use two camera to run ego-planner? one provides depth image,another provides position information?

001zwzwzw, 2022-04-09:
for example D435i and T265?

bigsuperZZZX, 2022-04-09:
We use D435 only with vins which positions our drones.

001zwzwzw, 2022-04-09:
yea! I konw ! Thank you very much

001zwzwzw, 2022-04-09:
salute！thank you very mucn

closed 2022-04-25 by bigsuperZZZX