ZJU-FAST-Lab/ego-planner-swarm #33: Trajectory tracking

shubham-shahh, 2022-04-25:
Hi @bigsuperZZZX,
as mentioned in the ego swarm paper, for trajectory tracking, the following [paper](https://ieeexplore.ieee.org/document/5980409) is referred to. I was curious if the codebase is open source for the trajectory tracker?

thanks

bigsuperZZZX, 2022-04-25:
We use the controller modified from [this codebase](https://github.com/HKUST-Aerial-Robotics/Teach-Repeat-Replan/tree/experiment/onboard_computer/controller/n3ctrl)

shubham-shahh, 2022-04-25:
oh okay, I'll configure ego planner with this controller and post the issues here

thanks a lot for the help

shubham-shahh, 2022-04-25:
Hi @bigsuperZZZX,
I am trying to understand and use N3ctrl package, as far as I understand, It subscribes to the IMU, and PositionTarget type of message which has position, velocity, ACC, yaw setpoints and it converts it publishes control command in the form of sensor_msgs/Joy

in the "axes" array of the message I am curious about the units of roll, pitch, yaw/yaw rate, 

thanks