ZJU-FAST-Lab/ego-planner-swarm #35: When using your so3 simulator, I publish the command

Mealoore, 2022-07-06:
When using your so3 simulator, I publish the command
```
header: 
  seq: 21
  stamp: 
    secs: 0
    nsecs:         0
  frame_id: ''
position: 
  x: nan
  y: nan
  z: nan
velocity: 
  x: 0.0
  y: 0.0
  z: 0.0
acceleration: 
  x: 0.0
  y: 0.0
  z: 0.0
yaw: 2.0
yaw_dot: 0.0
kx: [0.0, 0.0, 0.0]
kv: [0.0, 0.0, 0.0]
trajectory_id: 0
trajectory_flag: 0
```
the height of uav is not correct, I think it should be at original point

bigsuperZZZX, 2022-07-15:
so3 simulator is not easy to use. Please check [this function](https://github.com/ZJU-FAST-Lab/ego-planner-swarm/blob/8da8a01e7d01a9977306aaf2922cf85a1a3de71e/src/uav_simulator/so3_control/src/SO3Control.cpp#L38) to see what it requires.

closed 2022-10-08 by bigsuperZZZX

Mealoore, 2022-10-10:
Sorry to bother you, could you provide the paper about the so3 simulator in your project, I want to learn more about it to use the simulator.Thanks a lot. And is there another appropriate simulator about uav which is can use in swarm, more than 50 uavs?

bigsuperZZZX, 2022-10-17:
The paper related to this controller is "Geometric tracking control of a quadrotor UAV on SE(3)". 
If CPU performance is the bottleneck, please replace the simulator and controller with  [this package](https://github.com/ZJU-FAST-Lab/ego-planner-swarm/tree/master/src/uav_simulator/fake_drone).