ZJU-FAST-Lab/ego-planner-swarm #40: 运行catkin_make -j1出错

041802120, 2022-09-27:
运行catkin_make -j1时，出现错误：
fatal error: multi_map_server/MultiOccupancyGrid.h: 没有那个文件或目录

bigsuperZZZX, 2022-10-08:
[src/uav_simulator/Utils/multi_map_server/src/multi_map_server/msg/_MultiOccupancyGrid.py](https://github.com/ZJU-FAST-Lab/ego-planner-swarm/blob/6fd2d6d7aa6bb82eb21edfb8a1168e7adec6a472/src/uav_simulator/Utils/multi_map_server/src/multi_map_server/msg/_MultiOccupancyGrid.py)
这个头文件来自于这个消息，确保它被优先编译了

041802120, 2022-10-08:
谢谢大佬

closed 2022-10-11 by Chai8226

Yiyuiii, 2022-10-31:
> [src/uav_simulator/Utils/multi_map_server/src/multi_map_server/msg/_MultiOccupancyGrid.py](https://github.com/ZJU-FAST-Lab/ego-planner-swarm/blob/6fd2d6d7aa6bb82eb21edfb8a1168e7adec6a472/src/uav_simulator/Utils/multi_map_server/src/multi_map_server/msg/_MultiOccupancyGrid.py) 这个头文件来自于这个消息，确保它被优先编译了

大佬们，我也遇到这个问题，请问该使用什么命令解决呢