ZJU-FAST-Lab/ego-planner-swarm #43: unknown catkin make error probably with Eigen

JaimeParker, 2022-10-24:
this happens while making `ego-planner-swarm/src/planner/traj_utils/include/traj_utils/polynomial_traj.h`
```shell
ego-planner-swarm/src/planner/traj_utils/include/traj_utils/polynomial_traj.h:237:107: error: no match for ‘operator=’ (operand types are ‘Eigen::internal::enable_if<true, Eigen::IndexedView<Eigen::Matrix<double, -1, -1>, double, double> >::type’ {aka ‘Eigen::IndexedView<Eigen::Matrix<double, -1, -1>, double, double>’} and ‘double’)
  237 |           mat_jerk(i, j) = i * (i - 1) * (i - 2) * j * (j - 1) * (j - 2) * pow(ts, i + j - 5) / (i + j - 5);
      |                                                                                                           ^
```
actually I don't see any wrong with `Eigen::MatrixXd`, everything seems correct
```shell
In file included from /usr/local/include/eigen3/Eigen/Core:273,
                 from /usr/local/include/eigen3/Eigen/Dense:1,
                 from /usr/local/include/eigen3/Eigen/Eigen:1,
                 from /home/hazyparker/projects/ego-planner-swarm/src/planner/plan_manage/include/plan_manage/ego_replan_fsm.h:4,
                 from /home/hazyparker/projects/ego-planner-swarm/src/planner/plan_manage/src/ego_replan_fsm.cpp:2:
/usr/local/include/eigen3/Eigen/src/Core/MatrixBase.h:150:14: note: candidate: ‘template<class OtherDerived> Derived& Eigen::MatrixBase<Derived>::operator=(const Eigen::EigenBase<OtherDerived>&) [with OtherDerived = OtherDerived; Derived = Eigen::IndexedView<Eigen::Matrix<double, -1, -1>, double, double>]’
  150 |     Derived& operator=(const EigenBase<OtherDerived>& other);
      |              ^~~~~~~~
/usr/local/include/eigen3/Eigen/src/Core/MatrixBase.h:150:14: note:   template argument deduction/substitution failed:
In file included from /home/hazyparker/projects/ego-planner-swarm/src/planner/traj_utils/include/traj_utils/plan_container.hpp:9,
                 from /home/hazyparker/projects/ego-planner-swarm/src/planner/bspline_opt/include/bspline_opt/bspline_optimizer.h:11,
                 from /home/hazyparker/projects/ego-planner-swarm/src/planner/plan_manage/include/plan_manage/ego_replan_fsm.h:14,
                 from /home/hazyparker/projects/ego-planner-swarm/src/planner/plan_manage/src/ego_replan_fsm.cpp:2:
/home/hazyparker/projects/ego-planner-swarm/src/planner/traj_utils/include/traj_utils/polynomial_traj.h:237:107: note:   mismatched types ‘const Eigen::EigenBase<Derived>’ and ‘double’
  237 |           mat_jerk(i, j) = i * (i - 1) * (i - 2) * j * (j - 1) * (j - 2) * pow(ts, i + j - 5) / (i + j - 5);
      |                                                                                                           ^
In file included from /usr/local/include/eigen3/Eigen/Core:273,
                 from /usr/local/include/eigen3/Eigen/Dense:1,
                 from /usr/local/include/eigen3/Eigen/Eigen:1,
                 from /home/hazyparker/projects/ego-planner-swarm/src/planner/plan_manage/include/plan_manage/ego_replan_fsm.h:4,
                 from /home/hazyparker/projects/ego-planner-swarm/src/planner/plan_manage/src/ego_replan_fsm.cpp:2:
/usr/local/include/eigen3/Eigen/src/Core/MatrixBase.h:146:14: note: candidate: ‘template<class OtherDerived> Derived& Eigen::MatrixBase<Derived>::operator=(const Eigen::DenseBase<OtherDerived>&) [with OtherDerived = OtherDerived; Derived = Eigen::IndexedView<Eigen::Matrix<double, -1, -1>, double, double>]’
  146 |     Derived& operator=(const DenseBase<OtherDerived>& other);
      |              ^~~~~~~~
/usr/local/include/eigen3/Eigen/src/Core/MatrixBase.h:146:14: note:   template argument deduction/substitution failed:
In file included from /home/hazyparker/projects/ego-planner-swarm/src/planner/traj_utils/include/traj_utils/plan_container.hpp:9,
                 from /home/hazyparker/projects/ego-planner-swarm/src/planner/bspline_opt/include/bspline_opt/bspline_optimizer.h:11,
                 from /home/hazyparker/projects/ego-planner-swarm/src/planner/plan_manage/include/plan_manage/ego_replan_fsm.h:14,
                 from /home/hazyparker/projects/ego-planner-swarm/src/planner/plan_manage/src/ego_replan_fsm.cpp:2:
/home/hazyparker/projects/ego-planner-swarm/src/planner/traj_utils/include/traj_utils/polynomial_traj.h:237:107: note:   mismatched types ‘const Eigen::DenseBase<Derived>’ and ‘double’
  237 |           mat_jerk(i, j) = i * (i - 1) * (i - 2) * j * (j - 1) * (j - 2) * pow(ts, i + j - 5) / (i + j - 5);
      |                                                                                                           ^
In file included from /usr/local/include/eigen3/Eigen/Core:282,
                 from /usr/local/include/eigen3/Eigen/Dense:1,
                 from /usr/local/include/eigen3/Eigen/Eigen:1,
                 from /home/hazyparker/projects/ego-planner-swarm/src/planner/plan_manage/include/plan_manage/ego_replan_fsm.h:4,
                 from /home/hazyparker/projects/ego-planner-swarm/src/planner/plan_manage/src/ego_replan_fsm.cpp:2:
/usr/local/include/eigen3/Eigen/src/Core/Assign.h:55:30: note: candidate: ‘Derived& Eigen::MatrixBase<Derived>::operator=(const Eigen::MatrixBase<Derived>&) [with Derived = Eigen::IndexedView<Eigen::Matrix<double, -1, -1>, double, double>]’
   55 | EIGEN_STRONG_INLINE Derived& MatrixBase<Derived>::operator=(const MatrixBase& other)
      |                              ^~~~~~~~~~~~~~~~~~~
/usr/local/include/eigen3/Eigen/src/Core/Assign.h:55:79: note:   no known conversion for argument 1 from ‘double’ to ‘const Eigen::MatrixBase<Eigen::IndexedView<Eigen::Matrix<double, -1, -1>, double, double> >&’
   55 | EIGEN_STRONG_INLINE Derived& MatrixBase<Derived>::operator=(const MatrixBase& other)
      |                                                             ~~~~~~~~~~~~~~~~~~^~~~~
make[2]: *** [planner/plan_manage/CMakeFiles/ego_planner_node.dir/build.make:76: planner/plan_manage/CMakeFiles/ego_planner_node.dir/src/ego_planner_node.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [planner/plan_manage/CMakeFiles/ego_planner_node.dir/build.make:104: planner/plan_manage/CMakeFiles/ego_planner_node.dir/src/planner_manager.cpp.o] Error 1
make[2]: *** [planner/plan_manage/CMakeFiles/ego_planner_node.dir/build.make:90: planner/plan_manage/CMakeFiles/ego_planner_node.dir/src/ego_replan_fsm.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:5277: planner/plan_manage/CMakeFiles/ego_planner_node.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
Invoking "make -j8 -l8" failed

```

popvan-ws, 2022-10-27:
same issue , try use eigen3.3.7 and ceres1.14.0

JaimeParker, 2022-10-30:
@popvan-ws I found that ceres version better to be 1.14, also with a lower eigen version, thx

closed 2022-10-30 by JaimeParker

Bruce-Si, 2023-02-16:
Same error. Then I tried eigen3.3.7 and ceres1.14.0, but got error: variable or field ‘it’ declared void for (typename C::const_iterator it = container.begin(); Then I tried eigen3.2.9, but got the same error... What's your version of eigen and ceres?