ZJU-FAST-Lab/ego-planner-swarm #44: OA problem for GPU version

0wh, 2022-10-28:
I am interested in the GPU version. I followed the instructions in `Use GPU or Not` and was able to build it with no error. However it seems the drones stopped avoiding any obstacle. They are only avoiding other drones, then fly straight forward despite of the obstacles in the map, see the attached picture. The CPU version works well, this problem only happens when I change ENABLE_CUDA to true. How should I solve it? Thanks!
[image]

bigsuperZZZX, 2023-01-29:
That's because you have set the wrong CUDA_NVCC_FLAGS as described in "Use GPU or Not" section. 
You can visualize a depth stream in rviz to see if the depth looks right.

hildebrandt-carl, 2023-03-13:
I can't find an official Nvidia page, however you can find the `CUDA_NVCC_FLAGS` using either [this website](https://arnon.dk/matching-sm-architectures-arch-and-gencode-for-various-nvidia-cards/) or [this website](https://www.myzhar.com/blog/tutorials/tutorial-nvidia-gpu-cuda-compute-capability/). For example, I have a RTX 3090, and you can see that I need to set my `-gencode` to `86`. Therefor I would set `-gencode arch=compute_86,code=sm_86;`.