ZJU-FAST-Lab/ego-planner-swarm #61: Subscribe odometry msg without velocity information

BohemianRhapsodyz, 2023-05-14:
Hello, thanks for the great work.

I notice that in  [this line](https://github.com/ZJU-FAST-Lab/ego-planner-swarm/blob/30de3c9e11e6a2e06d81561f9364f07ceba6774b/src/planner/plan_manage/src/ego_replan_fsm.cpp#LL231C31-L231C31) the velocity value from odometry msg is used. I wonder that if I have an odometry with position and orientation only (i.e. the msg->twist.twist.linear.x/y/z are always zero), will the planner still work?

XXLiu-HNU, 2025-01-11:
I encountered this problem when using the motion capture system. I tried to combine the two topics.

bigsuperZZZX, 2025-03-08:
I would recommend adding an EKF to estimate velocity. The velocity used in ego-planner does not need to be accurate.