ZJU-FAST-Lab/ego-planner-swarm #65: Missing: multi_map_server/MultiOccupancyGrid.h during building --> found in https://github.com/HKUST-Aerial-Robotics/Fast-Planner/tree/master/uav_simulator/Utils/multi_map_server

Twenkid, 2023-09-25:
Hi, you've created an amazing code base both in this project and your lab! Good job!
I tried to compile the ego-planer-swarm project, but either ways for building it fails with the same error, a header is missing and I can't find it in the folders either.
```
...
catkin_make -j1
...
catkin_make -DCMAKE_BUILD_TYPE=Release -j1
```

[image]

I find only Python files with that name and underscore, in multi_map_server.zip  etc.

_MultiOccupancyGrid.py

I found the files here: 

https://github.com/HKUST-Aerial-Robotics/Fast-Planner/tree/master/uav_simulator/Utils/multi_map_server

...and I copied them to the directory, the building continued and succeeded.

[image]

Regards

HauserDong, 2023-10-23:
Yes, you can also find the requested head files in the following link: https://github.com/HKUST-Aerial-Robotics/plan_utils/tree/master/multi_map_server/msg_gen/cpp/include/multi_map_server

lbboier, 2024-05-29:
Using "catkin_make"  or "ctrl +shift +B" instend of "catkin_make -j1"

zoythum, 2025-02-24:
@Twenkid what files did you copy? And where did you copy them? Thanks!

Twenkid, 2025-02-27:
Hi, see the image with the error message and the links in the messages above. .   .../Utils/multi_map_server ... All 4 header files, if they are missing.