ZJU-FAST-Lab/ego-planner-swarm #75: [drone_0_ego_planner_node-2] process has died

Lockwuud, 2025-01-24:
Hello, outstanding work! I have successfully run ego-planner on my host machine. However, I am encountering an issue when running ego-planner-swarm: after successfully completing `catkin_make` as described in the README, I executed `roslaunch ego_planner rviz.launch` and `roslaunch ego_planner swarm.launch` in sequence. Unfortunately, I received error messages indicating that the processes for all 10 drone nodes have crashed. The specific error messages are as follows:

[image]

I am using an x86 platform with hardware specifications including a 13th Gen Intel i9 processor and an RTX 40 series GPU.

I look forward to receiving your effective solution. Thank you!

closed 2025-03-26 by Lockwuud

iceteen007, 2025-08-28:
wa

Lockwuud, 2025-08-28:
> I use Ubuntu 22.04+ros1noetic, which is very difficult. I installed it from source code and used the onboard computer of aipro.

I use Ubuntu 20.04 and ROS1 noetic.

iceteen007, 2025-08-28:
离谱死了，解决了两天，没解决掉

zsyeh, 2025-08-28:
国内

Lockwuud, 2025-08-28:
> 离谱死了，解决了两天，没解决掉

我当时有两个一样的上位机，同样的环境，把代码从一个能跑的上位机里直接拷贝到另一个上位机里，另一个就是遇到这个问题，我也不知道啥问题，干脆重装系统，结果还真解决了

iceteen007, 2025-08-28:
重装系统，又要装一堆东西
因为我这个版本不匹配

iceteen007, 2025-08-28:
费了好大的功夫，把所有的都通过编译了
离谱，遇到这个问题

zsyeh, 2025-08-28:
> 费了好大的功夫，把所有的都通过编译了 离谱，遇到这个问题

请问您的aipro是只提供22的发行版吗

iceteen007, 2025-08-28:
是这样的

Lockwuud, 2025-08-28:
我直接用的x86平台

zsyeh, 2025-08-28:
> 要不然谁这么头铁，拿着22.04的去用ros1 呀。无敌啦

您可以尝试 ros2/chroot ros1的根文件系统/docker with host and privilege

iceteen007, 2025-08-28:
但是我记得用ducker，好像是没办法直接装桌面版的完整的ros1。

zsyeh, 2025-08-28:
> 但是我记得用ducker，好像是没办法直接装桌面版的完整的ros1。

使用x11管道拿出来就可以 我之前长期在24.04上chroot使用noetic

zsyeh, 2025-08-28:
> 啊，真的吗！！！

是的，如果您需要交流细节可以邮件交换联系方式

iceteen007, 2025-08-28:
> > 即使我强行换了ROS1的元，由于版本不同，22.04和20.04的架构不同，还是不行的
> 
> 并非架构不同 只是极少量系统api不同 主要区别在于根文件系统和仓库源不同

所以，额，有点听不懂啦，不过，难道那样我就不用手动编译了吗！

iceteen007, 2025-08-28:
1444131552这是qq

iceteen007, 2025-08-28:
> > 啊，真的吗！！！
> 
> 是的，如果您需要交流细节可以邮件交换联系方式

怎么给你啊，……不太会，我好像看不到你的邮箱