ZJU-FAST-Lab/ego-planner-swarm #76: terminate called after throwing an instance of 'std::runtime_error'

zixie007, 2025-03-14:
作者，你好！我以真实实验下的里程计和点云话题替换ros2版本single_run_in_sim.launch里相应话题，此时可以正常运行。但当我以模式1（/move_base_simple/goal）重复发送目标点 或者 飞机碰到膨胀后的障碍物时，会报下面的错。我也试着ros2 node list检查已启动的节点，确实只有一个ego节点。
[ego_planner_node-1] terminate called after throwing an instance of 'std::runtime_error'
[ego_planner_node-1]   what():  Node '/drone_0_ego_planner_node' has already been added to an executor.
[ERROR] [ego_planner_node-1]: process has died [pid 9758, exit code -6, cmd '/home/khadas/ego-planner-swarm/install/ego_planner/lib/ego_planner/ego_planner_node --ros-args -r __node:=drone_0_ego_planner_node --params-file /tmp/launch_params_gw4l2890 --params-file /tmp/launch_params_78_bsalk --params-file /tmp/launch_params_o_u_lopj --params-file /tmp/launch_params_diifosmz --params-file /tmp/launch_params_4xlpfr1k --params-file /tmp/launch_params_hvqc6nrs --params-file /tmp/launch_params_uq30qw0r --params-file /tmp/launch_params_v4tbqqh7 --params-file /tmp/launch_params_yauskizt --params-file /tmp/launch_params_rbs1n38o --params-file /tmp/launch_params_oxwp8s26 --params-file /tmp/launch_params__7_5hsff --params-file /tmp/launch_params_p6o34ut1 --params-file /tmp/launch_params_crr0qtl9 --params-file /tmp/launch_params_o2sh47z5 --params-file /tmp/launch_params_k9t99jwg --params-file /tmp/launch_params_aagt8ebc --params-file /tmp/launch_params_2zsbnqao --params-file /tmp/launch_params_jf5xloqo --params-file /tmp/launch_params_tia4j0tw --params-file /tmp/launch_params_4jg_8i72 --params-file /tmp/launch_params_36zlxvcy --params-file /tmp/launch_params_tkh61_t9 --params-file /tmp/launch_params_blf12uc2 --params-file /tmp/launch_params_5diea0c3 --params-file /tmp/launch_params_tnljhpqb --params-file /tmp/launch_params_sq03nu8_ --params-file /tmp/launch_params_d5rgwv0z --params-file /tmp/launch_params_q6eezzdm --params-file /tmp/launch_params_9yto1e6y --params-file /tmp/launch_params_4nh1aehu --params-file /tmp/launch_params_t2q9g10n --params-file /tmp/launch_params_jkwpobx1 --params-file /tmp/launch_params_y8dvp2rp --params-file /tmp/launch_params_jgwvell3 --params-file /tmp/launch_params_wckygvcn --params-file /tmp/launch_params_kow9g7w5 --params-file /tmp/launch_params_skqnucv6 --params-file /tmp/launch_params_pi9zsgsm --params-file /tmp/launch_params_6cotl_cx --params-file /tmp/launch_params_zz2_14ip --params-file /tmp/launch_params_aqgedq50 --params-file /tmp/launch_params_ovehrjcu --params-file /tmp/launch_params_squkw6vo --params-file /tmp/launch_params_f01zxd_m --params-file /tmp/launch_params_ulumhj_r --params-file /tmp/launch_params_hjdbjl9e --params-file /tmp/launch_params_oy6syjn1 --params-file /tmp/launch_params_rvt3ujty --params-file /tmp/launch_params_bqt0pwbl --params-file /tmp/launch_params_g77pcyot --params-file /tmp/launch_params_5kx7rvlb --params-file /tmp/launch_params_gfag9mgl --params-file /tmp/launch_params_25ezk3mi --params-file /tmp/launch_params_pelhrjqm --params-file /tmp/launch_params_e571e735 --params-file /tmp/launch_params_i88j7b2h --params-file /tmp/launch_params_5df8iwcz --params-file /tmp/launch_params_86gw7f1o --params-file /tmp/launch_params_4dqmh8rm --params-file /tmp/launch_params_4pc83qwg --params-file /tmp/launch_params_5l13oucq --params-file /tmp/launch_params_mctyv7a4 --params-file /tmp/launch_params_1w0dmetw --params-file /tmp/launch_params_55lgcsaq --params-file /tmp/launch_params_tleb0eb1 --params-file /tmp/launch_params_vclyg3rq --params-file /tmp/launch_params_rqpiuv46 --params-file /tmp/launch_params_4fly9ch5 --params-file /tmp/launch_params_xj2nav41 --params-file /tmp/launch_params_0uqkc0fx --params-file /tmp/launch_params_nm7hnvas --params-file /tmp/launch_params_73ulcogp --params-file /tmp/launch_params_y30rt1oi --params-file /tmp/launch_params_i94s23hz --params-file /tmp/launch_params_wl866yv9 --params-file /tmp/launch_params_2esovfkj --params-file /tmp/launch_params_rpyfe8sl --params-file /tmp/launch_params_xhbefwin -r odom_world:=/mavros/odometry/out -r planning/bspline:=drone_0_planning/bspline -r planning/data_display:=drone_0_planning/data_display -r planning/broadcast_bspline_from_planner:=/broadcast_bspline -r planning/broadcast_bspline_to_planner:=/broadcast_bspline -r goal_point:=drone_0_plan_vis/goal_point -r global_list:=drone_0_plan_vis/global_list -r init_list:=drone_0_plan_vis/init_list -r optimal_list:=drone_0_plan_vis/optimal_list -r a_star_list:=drone_0_plan_vis/a_star_list -r grid_map/odom:=/mavros/odometry/out -r grid_map/cloud:=/out/cld/world -r grid_map/pose:=drone_0_camera_pose -r grid_map/depth:=drone_0_depth_image -r grid_map/occupancy_inflate:=drone_0_grid/grid_map/occupancy_inflate'].

YuyuYueuyue, 2025-06-10:
一样的 我也是

auto-lcx, 2025-12-30:
[ego_planner_node-1] Triggered!
[ego_planner_node-1] terminate called after throwing an instance of 'std::bad_alloc'
[ego_planner_node-1]   what():  std::bad_alloc
[ERROR] [ego_planner_node-1]: process has died [pid 21913, exit code -6, cmd '/home/lcx/drone_ws/install/ego_planner/lib/ego_planner/ego_planner_node --ros-args -r __node:=ego_planner_node --params-file /tmp/launch_params_a37xx__c --params-file /tmp/launch_params_w4v91f76 --params-file /tmp/launch_params_yseuk4b0 --params-file /tmp/launch_params_j8auzfi3 --params-file /tmp/launch_params_fwumfs5f --params-file /tmp/launch_params_huu45fpg --params-file /tmp/launch_params_fu0n4odj --params-file /tmp/launch_params_yo2meizg --params-file /tmp/launch_params_y7fx1s7m --params-file /tmp/launch_params_1nty6ys6 --params-file /tmp/launch_params_jcqpes40 --params-file /tmp/launch_params_wcaia8u7 --params-file /tmp/launch_params_0doiuquy --params-file /tmp/launch_params_9zi1q2l3 --params-file /tmp/launch_params_nm5nb97z --params-file /tmp/launch_params_ed2_sahp --params-file /tmp/launch_params_n1pnbds4 --params-file /tmp/launch_params_8ibfe8yk --params-file /tmp/launch_params_7_74i1wj --params-file /tmp/launch_params_euhhys9t --params-file /tmp/launch_params_7_x7s8hq --params-file /tmp/launch_params_qt7v49mu --params-file /tmp/launch_params_3q74d4o2 --params-file /tmp/launch_params_r6f6sdw6 --params-file /tmp/launch_params_aikj_rnu --params-file /tmp/launch_params_eiphue0g --params-file /tmp/launch_params_upd8mpu7 --params-file /tmp/launch_params_etsd2aoo --params-file /tmp/launch_params_5gk9y2t0 --params-file /tmp/launch_params_3_mcablw --params-file /tmp/launch_params_2tcrvgvi --params-file /tmp/launch_params_t90zam0p --params-file /tmp/launch_params_ip1m77sc --params-file /tmp/launch_params_8_1mgc6v -r odom:=/Odometry -r goal:=/ego_planner/goal -r traj_start_trigger:=/ego_planner/trigger -r pointcloud:=/livox/lidar/pointcloud -r planning/pos_cmd:=/ego_planner/position_cmd'].

我也是px4终端给了飞行命令触发后报错,模式1: MANUAL_TARGET, 2: PRESET_TARGET报错都是上面这个问题

YuyuYueuyue, 2026-01-15:
@bigsuperZZZX 救救救我 大佬5555

BillCai06, 2026-10-29:
The std::bad_alloc above may be an out-of-bounds read in reboundReplan: when a replan runs after the local trajectory has ended, segment_point is empty and segment_point[size()-2] is read; the garbage arc length then makes the resampling loop allocate without bound. Just add the pull requires