ZJU-FAST-Lab/ego-planner-swarm #77: Report a ROS2 build BUG: [cv_bridge] No such file or directory

andy-zhuo-02, 2025-03-22:
Ubuntu 24.04 LTS
ROS2： jazzy

在进行编译时会遇到类似如下报错：
```
fatal error: cv_bridge/cv_bridge.h: No such file or directory
    6 | #include <cv_bridge/cv_bridge.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```
但是剑法发现确实已经安装了cv_bridge库。

检查发现将include语句
`#include <cv_bridge/cv_bridge.h>`
修改为
`#include <cv_bridge/cv_bridge.hpp>`
之后，可以正常编译

andy-zhuo-02, 2025-03-22:
另外在我的环境中运行的时候，没有修改DDS，但并未遇见README中提及的程序很卡的问题

bigsuperZZZX, 2025-03-24:
了解了，多谢

closed 2025-03-24 by bigsuperZZZX

EzioPeter, 2026-01-14:
> > [@andy-zhuo-02](https://github.com/andy-zhuo-02) 想请问一下，jazzy的cv_bridge需要如何安装，ros-perception似乎目前没有开放适用于jazzy的cv_bridge：https://github.com/ros-perception/vision_opencv
> 
> 我查询我Ubuntu24下的cv-bridge版本：
> 
> ```
> dpkg -l | grep -E 'ros-.*-cv-bridge|cv-bridge'
> ii  ros-jazzy-cv-bridge                              4.1.0-1noble.20251025.091614             amd64        This contains CvBridge, which converts between ROS2 Image messages and OpenCV images.
> ```
> 
> 应该直接尝试`sudo apt install ros-jazzy-cv-bridge`就可以

我也查询了下我ubuntu 24系统的ros-jazzy-cv-bridge，也是和你的同一效果同一版本，但是apt install后并对整个系统搜索cv_bridge.h，并不能在/opt/ros/jazzy下找到相应的cv_bridge.h，编译需要使用到cv_bridge时也是无法找到。

我目前在ubuntu 24下强行使用<https://github.com/ros-perception/vision_opencv>的humble版本的cv_bridge，能够正常编译，正常使用ego planner在ros2下的rviz仿真，但是还不清楚部署到真机上后会不会存在问题