ZJU-FAST-Lab/ego-planner-swarm #84: Real drone - LiDAR instead of depth camera

aktamaa, 2025-07-18:
First of all, thank you so much to the author for sharing this great works!

I am currently trying to integrate ego planner to my drone which uses LiDAR instead of depth camera. I did implement it with depth camera and everything was working smoothly. But once I use LiDAR, it seems that the transformation of the grid map is not correct. I am using the combination of point cloud and odometry data. The modification that I made in the launch file was like this, in the case of depth camera.

```
'odom_topic': '/uav0/odometry'
'depth_topic': '/uav0/depth'
'grid_map/pose_type': '2'
'grd_map/frame_id': 'world'
```

What I did is just using the parameters settings that I used for the depth camera, following your fast drone 250 tutorial and just change the topic from depth to cloud, like this.

```
'odom_topic': '/uav0/odometry'
'cloud_topic': '/uav0/livox/lidar'
'grid_map/pose_type': '2'
'grd_map/frame_id': 'world'
```

The result was that the map is rotating and I am pretty sure that the drone will not fly correctly avoiding the obstacles with the current grid map condition. I tried also to follow your suggestion from the issue [#19](https://github.com/ZJU-FAST-Lab/ego-planner-swarm/issues/19), but I am not sure that I did it correctly or not, as the map still looks the same as before I made changes in the `grid_map.cpp`. Currently the lidar point cloud is published on the camera frame, not the sensor frame. Please kindly share your suggestion on this one. Again, thank you so much!

https://github.com/user-attachments/assets/83fc336e-58a3-44e4-aa74-b0e4b358ff2a

modified odomCallback:
```
void GridMap::odomCallback(const nav_msgs::msg::Odometry::SharedPtr odom)
{
  if (md_.has_first_depth_)
    return;

  md_.camera_pos_(0) = odom->pose.pose.position.x;
  md_.camera_pos_(1) = odom->pose.pose.position.y;
  md_.camera_pos_(2) = odom->pose.pose.position.z;

  Eigen::Quaterniond body_q = Eigen::Quaterniond(odom->pose.pose.orientation.w,
                                                 odom->pose.pose.orientation.x,
                                                 odom->pose.pose.orientation.y,
                                                 odom->pose.pose.orientation.z);
  md_.camera_r_m_ = body_q.toRotationMatrix();

  md_.has_odom_ = true;
}
```

modified cloudCallback:
```
...

 pt = latest_cloud.points[i];
 p3d(0) = pt.x, p3d(1) = pt.y, p3d(2) = pt.z;
 p3d = md_.camera_r_m_ * p3d + md_.camera_pos_;

...
```

aktamaa, 2025-07-22:
Updating this, the problem was just that I didn't adopt the transformed point cloud, the `p3d`, to the point cloud inflation part haha. So the final cloudCallback will be as follows, in case that someone is looking for the solution to the same problem like mine, to integrate ego swarm to a lidar-based auto nav drone. The way I am doing the transformation is correct I think, so anyone can also try to do the same, by modifying the odomCallback function like above. And again, big thanks to the author!

```
for (size_t i = 0; i < latest_cloud.points.size(); ++i)
  {
    pt = latest_cloud.points[i];
    p3d(0) = pt.x, p3d(1) = pt.y, p3d(2) = pt.z;
    p3d = md_.camera_r_m_ * p3d + md_.camera_pos_;

    /* point inside update range */
    Eigen::Vector3d devi = p3d - md_.camera_pos_;
    Eigen::Vector3i inf_pt;

    if (fabs(devi(0)) < mp_.local_update_range_(0) && fabs(devi(1)) < mp_.local_update_range_(1) &&
        fabs(devi(2)) < mp_.local_update_range_(2))
    {

      /* inflate the point */
      // 点云膨胀
      for (int x = -inf_step; x <= inf_step; ++x)
        for (int y = -inf_step; y <= inf_step; ++y)
          for (int z = -inf_step_z; z <= inf_step_z; ++z)
          {

            p3d_inf(0) = p3d(0) + x * mp_.resolution_;
            p3d_inf(1) = p3d(1) + y * mp_.resolution_;
            p3d_inf(2) = p3d(2) + z * mp_.resolution_;

            max_x = max(max_x, p3d_inf(0));
            max_y = max(max_y, p3d_inf(1));
            max_z = max(max_z, p3d_inf(2));

            min_x = min(min_x, p3d_inf(0));
            min_y = min(min_y, p3d_inf(1));
            min_z = min(min_z, p3d_inf(2));

            posToIndex(p3d_inf, inf_pt);

            if (!isInMap(inf_pt))
              continue;

            int idx_inf = toAddress(inf_pt);

            md_.occupancy_buffer_inflate_[idx_inf] = 1;
          }
    }
  }
```

closed 2025-07-22 by aktamaa

SomehowL, 2025-11-25:
@aktamaa Hi, have you tried ROS2 version with Lidar?

aktamaa, 2025-12-16:
@SomehowL To make the drone be able to follow the trajectory correctly, unfortunately there are many things to consider actually like the drone capability (thrust to wieght ratio, PID settings, etc), localization, computing device, etc. For the localization, in my case, I use LIO which provides high accuracy state estimation and I highly recommend it if you have the lidar already. There are some great open sourced packages you can use for it such as [GLIM](https://github.com/koide3/glim.git) and [FAST-LIO](https://github.com/hku-mars/FAST_LIO.git). It is possible to use GPS (not RTK) but I don't recommend it. For the GNSS-based localization case, I use RTK previously instead of just GPS to achieve centimeter accuracy

aktamaa, 2026-02-02:
@auto-lcx hello. yes I did, with ROS2, for both single and swarm drones